rmodbus/rmodbus #34: EOFError with RTUClient

mtuckerb, 2015-03-11:
I'm using the RTUClient module to connect to a MODBus device that is attached via serial->usb device (Keyspan) to my Mac (OSX 10.10). I try to read the holding registers and get  a holding error like [this](https://gist.github.com/mtuckerb/1d0ff136c6453688eefd). I can access the registers through a general purpose MODBus gui. Is this a bug, or am I doing something wrong?

flipback, 2015-03-12:
Thanks for the report! It seems I've found the solution of the problem. But it needs to be tested. Could you check my bugfix from the repository, please? Before I'll publish gem. I can't do it on my own, because I haven't got any RTU devices at hand.

mtuckerb, 2015-03-12:
That seems to work like a charm! Thank you.

Tucker Bradford
http://forgeover.com

flipback, 2015-03-12:
Excellent! I've released bugfix gem. So you can update yours. 
Thanks a lot for your help!

closed 2015-03-12 by flipback