rmodbus/rmodbus #41: Nil Error: msg.read(1) seems to return nil

kaiserprogrammer, 2015-08-31:
Hi, i use rmodbus in production and occasionally i see this error in the logs. Maybe you know what to do with it or what it means.
```
#<TypeError: no implicit conversion of nil into String> no implicit conversion of nil into String 
"/var/lib/gems/2.2.0/gems/rmodbus-1.2.6/lib/rmodbus/rtu.rb:34:in `+'",
"/var/lib/gems/2.2.0/gems/rmodbus-1.2.6/lib/rmodbus/rtu.rb:34:in `read_rtu_response'",
"/var/lib/gems/2.2.0/gems/rmodbus-1.2.6/lib/rmodbus/rtu.rb:63:in `read_rtu_pdu'",
"/var/lib/gems/2.2.0/gems/rmodbus-1.2.6/lib/rmodbus/rtu_slave.rb:39:in `read_pdu'",
"/var/lib/gems/2.2.0/gems/rmodbus-1.2.6/lib/rmodbus/slave.rb:248:in `block in query'",
"/usr/lib/ruby/2.2.0/timeout.rb:88:in `block in timeout'",
"/usr/lib/ruby/2.2.0/timeout.rb:98:in `call'",
"/usr/lib/ruby/2.2.0/timeout.rb:98:in `timeout'",
"/usr/lib/ruby/2.2.0/timeout.rb:124:in `timeout'",
"/var/lib/gems/2.2.0/gems/rmodbus-1.2.6/lib/rmodbus/slave.rb:246:in `query'",
"/var/lib/gems/2.2.0/gems/rmodbus-1.2.6/lib/rmodbus/slave.rb:177:in `read_holding_registers'",
"/var/lib/gems/2.2.0/gems/rmodbus-1.2.6/lib/rmodbus/proxy.rb:26:in `[]'",
```

flipback, 2015-09-01:
Thanks for the report! Do you use a serial port or RTU via TCP? And if it is a serial port, is it a "real" port or an USB/COM converter?

kaiserprogrammer, 2015-09-01:
It's an USB/COM converter, there are 7 running in parallel.

flipback, 2015-09-01:
I see. It might be the matter. I'll change the library so that it doesn't throw an exception when it's received nil but writes a warning message in the log and carries on working.

ccutrer, 2024-08-20:
Version 2.x properly handles serial ports returning nil (technically EOF, but not really, since a serial port can always receive more data later)

closed 2024-08-20 by ccutrer