rmodbus/rmodbus #46: use rubyserial instead of serialport

kaiserprogrammer, 2016-02-09:
Can we change rmodbus to use rubyserial instead of serialport, since it's currently leaking memory on windows and it does not work with jruby.

Changes needed:
  - rubyserial has no flush
  - rubyserial has non-blocking read (easily solved by a loop, timeout should work there also)

flipback, 2016-02-09:
It's a good idea, but I'm afraid that I don't have enough time to implement it. Moreover I'm not sure that I will be able to support the project in the future. So if you are enthusiastic about the project you can do it by yourself. I'd be very happy if anybody carries on developing rmodbus.

fernandoluizao, 2016-09-16:
Hi,
Do you still have interest in this?

ccutrer, 2024-08-20:
The 2.x version uses the ccutrer-serialport gem, which only uses FFI, not an extension, so should work from JRuby.

closed 2024-08-20 by ccutrer