rmodbus/rmodbus #47: Very frequent timeouts (solved by monkey patching)

Insood, 2016-02-25:
Hello,

My setup: 
OS: Windows 10 Home
Ruby: 2.1.5p273
rmodbus: 1.2.7
PLC: Click PLC C0-00DD1-D
USB: EA-MG-PGM-CBL (Prolific PL-2303 TA)

Issue: code occasionally (!) kept getting stuck in the read_rtu_response io.read() loop. I tried a large number of things to figure out what was going on, but eventually settled on the following:

```
module ModBus
	module RTU
	private
		def send_rtu_pdu(pdu)
			msg = @uid.chr + pdu
			msg << crc16(msg).to_word
			clean_input_buff
			sleep 0.005
			@io.write msg
			sleep 0.005
		end
	end
end
```

Unfortunately I don't fully understand why the above was required. It may be that my PLC is not able to switch between read/write fast enough?

fernandoluizao, 2016-09-16:
Hi,
Can you provide a sample of your code where it hangs?

fernandoluizao, 2018-08-23:
@naveedkakal I think Errno:EADDRINUSE is only raised when a server tries to bind to some port. The client just establishes the connection. Can you show the code which is raising the exception? The backtrace will also be helpful.

ccutrer, 2024-08-20:
Version 2.x reworks the read loop. Since this issue is so old, I'm going to assume it's fixed unless it's re-opened.

closed 2024-08-20 by ccutrer