rmodbus/rmodbus #51: Read 32 bit floating point holding registers

gfoiani, 2016-10-17:
Hi all,

I'm trying to read a sequence of 32 bit floating point holding registers.
Can someone give me an example of how to do that?

I can read those registers but they are represented as 16 bit registers...

fernandoluizao, 2016-10-17:
You can call to_32f like this:

```
slave.holding_registers[0..1].to_32f
=> [1.345676]
```

You will get an array of floats, since you can read more than 2 registers at once, to_32f will convert them all as long as they are multiples of 2.

gfoiani, 2016-10-17:
Thanks @fernandoluizao,

I tried with `.to_32f` but I get a wrong value. The real value set on modbus slave is 0.5 but calling .to_32f I get `[2.260014163263065e-41]`.

Any idea about the reason?

fernandoluizao, 2016-10-17:
Array#to_32f assumes big endian, your device seems to use little endian.

See https://github.com/rmodbus/rmodbus/pull/50/files for an alternative.

I'm thinking about remove those extensions, and implement https://github.com/rmodbus/rmodbus/issues/19, as I think this conversions are outside the scope of a modbus library.

But for now, you can monkeypatch Array (I know, I know...)

gfoiani, 2016-10-18:
Thanks @fernandoluizao,

the Array patch works. For the moment, the solution may be acceptable.

closed 2016-10-18 by gfoiani