rmodbus/rmodbus #55: Timeouts When Writing to Slave

stevehayenergy, 2018-04-26:
I am getting timeouts when writing to a slave very aggressively. I have 30 or so threads all writing values to the same register, and the rmodbus slave very reliably times out. I think there is probably a need for a Mutex somewhere, because I think the io streams are stepping on each other.

If I slow down the sequence of writes, it gets more reliable. I think the problem occurs when more than one connection is writing to the slave at the time time. This happens whether the threads are writing to the same register or different registers.

fernandoluizao, 2018-04-27:
Which protocol are you using? I will review the code to find possible solutions.

fernandoluizao, 2018-05-10:
Sorry for the delay, I've been a little busy with another project.

Could you try the **mutex** branch to see if the problem has been solved? If you can, please make some benchmarks, and let me know if there was an impact on the performance.

jhadeepakkumar14, 2018-08-29:
I am also facing the same issue.

fernandoluizao, 2018-08-29:
@jhadeepakkumar14 can you try with [this](https://github.com/rmodbus/rmodbus/tree/mutex) branch please?

ccutrer, 2024-08-20:
This is a duplicate of #16.

closed 2024-08-20 by ccutrer