rmodbus/rmodbus #57: to_32i is little-endian, not big endian

nemecsek69, 2019-09-23:
`to_32i` in ext.rb is defined as little endian even if the comment reads _big endian_.

I met this problem reading two consecutive words in a U32 register:
```
reg[18458] = 5
reg[18459] = 28000
``` 

Being big-endian the U32 should have been 
5\*65536 + 28000 = 355680, but it returns instead 28000\*65536 + 5 = 1835008005.

I am not sure if I am missing anything, so I prefer not to request for a Pull. 

The code in ext.rb should be changed to: 
```ruby
# Given an array of 16bit Fixnum, we turn it into 32bit Float in big-endian order, halving the size
def to_32i
  raise "Array requires an even number of elements to pack to 32bits: was #{self.size}" unless self.size.even?
  self.each_slice(2).map { |(msb, lsb)| [msb, lsb].pack('n*').unpack('N')[0] }
end
```
`to_32i_le` could be also added (it is the original code):
```ruby
# Given an array of 16bit Fixnum, we turn it into 32bit Float in little-endian order, halving the size
def to_32i_le
  raise "Array requires an even number of elements to pack to 32bits: was #{self.size}" unless self.size.even?
  self.each_slice(2).map { |(lsb, msb)| [msb, lsb].pack('n*').unpack('N')[0] }
end
```

fernandoluizao, 2019-09-23:
Indeed, it's wrong since [it was introduced](https://github.com/rmodbus/rmodbus/commit/8673eac3835eb383ff1f5d72cabc57f0bf904681#diff-edbe6d24ba4a579b65069894f83fb160R56).

kreynolds, 2019-09-23:
Yep, I biffed the comment on that, good catch.

fernandoluizao, 2019-09-26:
Since it's a breaking change, I'm thinking in prepare a 1.0 relase for next week.
I'll need to review other things too...

closed 2024-08-20 by ccutrer